donovan6000/iMe #1: unable to connect | Mac

sfiamma, 2016-05-14:
I'm trying to upload the iMe firmware with M3D Manager on my Macbook, but i'm not able to connect.
With M3D original software is all ok.
Any suggestions ?

scallviz, 2016-05-16:
Looks like it might be related to a missing piece of pyserial:
M3D\ Manager.app/Contents/MacOS/M3D\ Manager
Traceback (most recent call last):
  File "<string>", line 1, in <module>
ImportError: No module named serial.tools.list_ports

and 
python -m serial.tools.list_ports
/usr/bin/python: No module named serial.tools

donovan6000, 2016-05-16:
Looks like I'll have to use OS X's API for finding the serial ports instead of using Python as a workaround. Both iMe and M3D Manager are still early in development, so I expected there to be a few problems.

donovan6000, 2016-05-18:
It should work on OS X now.

JulienDeray, 2016-05-26:
Still not connecting for me. I basically just have to load the firmware with M3D Manager and hit "print", right ?

donovan6000, 2016-05-26:
Every time the printer powers on it'll be in bootloader mode, so you need to use M3D Manager to switch the printer into firmware mode before you can use the printer with 3D printing software. If your using Cura 15.04, then the autodetecting serial port feature doesn't work very well, so you'll need to set it manually in the machine settings menu. Also make sure the baudrate is set to 115200 in the machine settings too.

closed 2016-06-07 by donovan6000