donovan6000/iMe #3: Movement in the X-Y axis intermittent

qwerty8224, 2016-05-20:


donovan6000, 2016-05-20:
This is due to how the bed leveling compensation works. It breaks up every movement into 2mm sections and determines the Z value that it needs to move to for that section to compensate for the bed's leveling. The Z motors moves much slower than the XY motors, so the XY movements are slowed down to guarantee that all motors stop at the exact same time. The amount that an XY movement is slowed down is proportional to the distance of the Z movement, so the irregularities that occur during XY movements are a result of the Z value for each section differing.

The firmware makes it so that all XYZE movements end at the same time, so the intermittent movements shouldn't be negatively affecting print quality.

qwerty8224, 2016-05-20:
Thank you and I thought that that was a Problem in version 6 has not been solved completely. Thank you for your work will testit more)))
The quality of the print on the face worse than the version MOD20160404

closed 2016-05-23 by qwerty8224