donovan6000/iMe #11: Bed Height Offset

qwerty8224, 2016-06-02:
Incorrect indicators

donovan6000, 2016-06-02:
What's wrong with the bed height offset? iMe should be automatically adjusting its internal Z value whenever the bed height offset changes.

For example, lets say your current Z is at 1 and the bed height offset is at 0.
`M114` = Z:1

Setting the bed height offset to 1 means that the entire bed has raised by 1mm. So to account for this change, iMe lowers its internal Z value by 1mm which results in Z being 0.
`M114` = Z:0

That means that iMe now thinks that the extruder is 1mm lower than it previously was, which is what would happen if the bed was physically raised by 1mm.

iMe is only ever changing its internal Z value and never physically changing the extruder's position when it accounts for changes to the bed height offset. It also does this whenever the bed corner offsets or bed corner orientations settings are changed.

qwerty8224, 2016-06-02:
Everything is fine. This is my problem bed.(((

donovan6000, 2016-06-03:
Yes, the bed height offset affects every layer. If it's set to 5mm, then the entire model will print 5mm higher. It's not a very useful setting since it should be 0 if your bed is calibrated correctly. I only implement it in M3D Fio and iMe since M3D's official software implements it.

qwerty8224, 2016-06-03:
Thank you. How do you polish bed?

donovan6000, 2016-06-03:
You can calibrate the bed using M3D Fio's 'Run complete bed calibration' button in the Calibration section of OctoPrint's Control tab.

Try [iMe's latest devel version](https://github.com/donovan6000/iMe/blob/devel/iMe%201900000010.hex?raw=true) and let me know if there's still issues with heating in Cura 14.04.05.

qwerty8224, 2016-06-03:
Cura 15.04.05 IME .....010 It does not heat.
M3D v0.06 not correctly display win10

donovan6000, 2016-06-03:
I just tried it with Cura 15.04.05, and it's working fine for me. Cura's autodetecing serial and baud rate can fail at times, so you might have to manually specify the serial port and baud rate in Cura's Machine Settings menu.
[image]

The devel version of M3D Manager should look weird. The devel branch contains anything that I'm currently working on, so it'll frequently contain unfinished and unstable code.

qwerty8224, 2016-06-03:
With connecting all OK.

closed 2016-06-04 by qwerty8224