donovan6000/iMe #13: MatterHackers

qwerty8224, 2016-06-07:
Sorry that I write here !!!
Moderator Matternachkers asks  profiles that would add to program the printer M3D

qwerty8224, 2016-06-07:
Maybe it's wrong but I think that it is necessary consent Donovan6000 )))

donovan6000, 2016-06-07:
The profiles ra100 linked to are the latest ones that we've been working on, so they should be good.

You could also ask @stephenci for a MatterControl profile for the Micro 3D. [He's done some experimenting with getting them to work nicely together](https://www.youtube.com/watch?v=FXfaQPzQaWw).

stephenci, 2016-06-08:
I had the basic setting changed to suit M3D but not thoroughly yet. I'm waiting for the firmware that will fix stuttering issue and then I'll find time the setting some more and upload the profile. Would it help to upload profile here in this hub?

closed 2016-06-11 by qwerty8224