donovan6000/iMe #14: Motor idle pitching sound

stephenci, 2016-06-08:
I noticed after homing or moving the head around, the motor would emit noise higher than it would normally with m3D fio. Also M84 or hitting the "release" button doesn't not seem to do anything I was hoping it would turn the motor off and stop the noise.

qwerty8224, 2016-06-08:
I think to compare MCH and IME impossible. Since this is a completely different products)))

stephenci, 2016-06-08:
Not sure if I understand you entirely on the MCH IME... I merely reporting issue I encounter here that I think can be improved

jtaylor330, 2016-06-10:
I'll second the noise and release function not working. Also if I don't Home X before printing my print (in matter control) will start centered nearly over x80y80 approximately even though it goes through homing X and Y at the start of the print it still wants to shift over and forward.

donovan6000, 2016-08-02:
iMe V00.00.01.06's motor idle pitch should be similar to that of M3D's official firmware. Also the command to turn the motors off is M18 for both iMe and M3D's official firmware.

closed 2016-08-02 by donovan6000