donovan6000/iMe #16: Manager compilation for Raspberry Pi

th0ma5w, 2016-06-14:
Attached is a copy of the Makefile I used to compile the manager to run on a Pi v2 ... I also had to download the dev headers for libwxgtk3 as well. It compiles without error, loads and shows the GUI, but I  haven't used it to download the firmware yet. Mostly it removes -m64 and also the -static from the wx options. So making a self contained executable may take some work or may not be possible? I don't know.

[Makefile.txt](https://github.com/donovan6000/iMe/files/313254/Makefile.txt)

donovan6000, 2016-06-14:
I don't own a Raspberry Pi, so I don't plan on providing a version of M3D Manager that supports it since compiling and testing M3D Manager for a device that I don't have access to seems troublesome. I assume Linux users are comfortable with compiling M3D Manager if they need it for a specific platform, which is why I only provide a compiled Linux version of M3D Manager for x86-64 platforms. 

If you want to statically link wxWidgets into M3D Manager then you'll probably have to compile wxWidgets to allow that. Here's how I think you'd do that for an ARM platform like the Raspberry Pi.
```
svn checkout http://svn.wxwidgets.org/svn/wx/wxWidgets/trunk/ wxWidgets-svn
cd wxWidgets-svn
./configure --prefix=/usr --target=arm-linux-gnueabihf --build=arm-linux-gnueabihf --enable-unicode --with-opengl --enable-static --disable-shared --enable-monolithic CC=gcc CXX=g++ CFLAGS="-std=c99" CXXFLAGS="-std=c++11" OBJCXXFLAGS="-std=c++11" LDFLAGS="-static-libgcc -static-libstdc++"
make && sudo make install
```
It's not really possible to statically link GTK+ into an executable, so it's impossible to make the Linux versions of M3D Manager completely self contained.

th0ma5w, 2016-06-15:
No worries, and thanks for the tips! I guess I wouldn't call this an issue per se, I was able to go on with life, heh, thank you so much for your work with the M3D!!!!! Perhaps this can just stay out there as some helpful pointers in case anyone else messes with this. Thanks again!

closed 2016-06-15 by th0ma5w