donovan6000/iMe #17: How can I raise the current to the motor?

qwerty8224, 2016-06-14:


donovan6000, 2016-06-14:
The current for the [X](https://github.com/donovan6000/iMe/blob/master/motors.cpp#L60), [Y](https://github.com/donovan6000/iMe/blob/master/motors.cpp#L72), and [Z](https://github.com/donovan6000/iMe/blob/master/motors.cpp#L84) motors are all hard coded, so you can change them in iMe source code. The E motor current is stored in the [printer's EEPROM](https://github.com/donovan6000/iMe/blob/master/eeprom.h#L117-L118) at address 0x2E8 as an unsigned 16-bit number in little-endian format as milliamps, so you can change it by changing the values in the printer's EEPROM.

Printer's with a serial number that starts with BK15033001100, BK15040201050, BK15040301050, BK15040602050, BK15040801050, BK15040802100, GR15032702100, GR15033101100, GR15040601100, GR15040701100, OR15032701100, or SL15032601050 are only designed to handle a max E motor current of 500mA, so don't go beyond that if that applies to your printer.

qwerty8224, 2016-06-15:
Unfortunately I am not strong in the 16 digit system ((((( But thanks anyway )))

closed 2016-06-16 by qwerty8224