donovan6000/iMe #25: Backlash settings

neonode, 2016-07-22:
Hello Donovan, 

First I want to admit you did great job. Much much better from everything M3D LLC ever manage to produce. You bring new life to my printer. 

I have problem  that I cannot save backlash values. This force me to flash firmware every time I want to change those values back to stock firmware. I installed Octoprint but when I set the values it say saved but when I switch to M33 Manager it show the old values. I am using Version 00.00.12 of iME. Also it will be really handy as iME do not support the stock custom GCodes for change of Backlash and offsets if you could  add this functionality in M33 manager.

qwerty8224, 2016-07-22:
Download from the devel branch there this setting 
https://github.com/donovan6000/iMe/tree/devel

neonode, 2016-07-22:
Thanks for the answer. But please be more specific. I downloaded the M33 Manger from the link but it seems same and I can't find those settings and a way how to modify them.

donovan6000, 2016-07-23:
It's working when I try it. What operating system are you using? Can you take a screen shot of OctoPrint's terminal tab right after changing the backlash setting(s) and clicking save. It should look similar to this.
[image]
This is me changing the backlash X to 0.3201. `M618 S12 T4 P1050928166` is the command that's writing that value to the printer's EEPROM. M618 being the actual command, S12 being the [offset in the EEPROM](https://github.com/donovan6000/iMe/blob/master/eeprom.h#L19), T4 being the [number of bytes in EEPROM that we're writing to](https://github.com/donovan6000/iMe/blob/master/eeprom.h#L20), and P1050928166 being the bytes to write.

0.3201 as a [single-percision floating-point value in little endian in hexadecimal](http://gregstoll.dyndns.org/~gregstoll/floattohex/) is 0x3EA3E426, which is 1050928166 in decimal.

`M619 S12 T4

donovan6000, 2016-07-23:
Also, the devel branch now has [M33 Manager](https://github.com/donovan6000/iMe/tree/devel/M33%20Manager) with all its controls shown, so you can also use that change the backlash settings too.

neonode, 2016-07-23:
The new M33 Manager work awesome thanks a lot! 

Also just an idea:

I am calibrating my backlash with a micro gauge as I do with my CNC mill. But just I was thinking today you can do it based on the accelerometer that is builtin. Just make issue movement in one direction of 5mm than issue series of movement of 0.005 in opposite count how many times you issue movement without reading acceleration from the accelerometer and calculate the backlash. Because the assembly is really loosy for this device you will not get accurate reading. So do this in opposite direction and calculate again. Move printer head with 5mm in direction just to get different position of the belt/rack and do it again. Execute this at least 5 times and make equation of the  of those readings and you will get pretty accurate backlash settings for every direction. 

Thanks again for this add-on in M33 manager! it really help a lot!

Regards,
George

closed 2016-07-23 by neonode

donovan6000, 2016-07-24:
The accelerometer that the printer uses is too susceptible to noise to be able to calibrate the backlash. iMe only uses the accelerometer for homing the extruder and for calibrating Z0 and M3D's firmware only uses it for calibrating Z0 since it's not good e
ses it for calibrating Z0 since it's not good enough to accurately track the extruder's position.

I'd still like to see what OctoPrint's terminal tab shows when you try to save to backlash settings so that we can see if there's an issue with M33 Fio.

neonode, 2016-07-24:
It seems my installation is broken somehow. I can not calibrate or update anything.When I click on calibration nothing happen it start processing and this continue forever also when I click on other command  it say ok but nothing happen and nothing show up in terminal. Here is the error stack:

2016-07-24 21:45:44,914 - octoprint.server.util.sockjs - INFO - New connection from client: 127.0.0.1
2016-07-24 21:45:45,072 - octoprint.plugin - ERROR - Error while calling plugin m33fio
Traceback (most recent call last):
  File "C:\Users\Neonode\AppData\Roaming\Python\Python27\site-packages\octoprint-1.2.13-py2.7.egg\octoprint\plugin\__init__.py", line 204, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "C:\Users\Neonode\AppData\Roaming\Python\Python27\site-packages\octoprint_m33fio\__init__.py", line 4567, in on_event
    
  File "C:\Users\Neonode\AppData\Roaming\Python\Python27\site-packages\octoprint_m33fio\__init__.py", line 4427, in setFileLocations
    
  File "C:\Users\Neonode\AppData\Roaming\Python\Python27\site-packages\octoprint_m33fio\__init__.py", line 1195, in convertCuraToProfile
    # Remove temporary file
WindowsError: [Error 32] The process cannot access the file because it is being used by another process: 'c:\\users\\neonode\\appdata\\local\\temp\\tmp7buejk'
2016-07-24 21:45:47,394 - octoprint.plugins.softwareupdate - INFO - Saved version cache to disk
2016-07-24 21:45:47,869 - octoprint.plugins.announcements - INFO - Loaded channel _important from http://octoprint.org/feeds/important.xml in 0.46s
2016-07-24 21:46:26,944 - octoprint.plugins.softwareupdate - INFO - Starting update of m33fio to V1.2...
2016-07-24 21:46:45,953 - octoprint.plugins.softwareupdate - INFO - Update of m33fio to V1.2 successful!
2016-07-24 21:46:45,966 - octoprint.plugins.softwareupdate - INFO - Saved version cache to disk
2016-07-24 21:48:04,944 - octoprint.server.util.sockjs - INFO - Client connection closed: 127.0.0.1
2016-07-24 21:48:06,257 - octoprint.server.util.sockjs - INFO - New connection from client: 127.0.0.1
2016-07-24 21:48:06,365 - octoprint.plugin - ERROR - Error while calling plugin m33fio
Traceback (most recent call last):
  File "C:\Users\Neonode\AppData\Roaming\Python\Python27\site-packages\octoprint-1.2.13-py2.7.egg\octoprint\plugin\__init__.py", line 204, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "C:\Users\Neonode\AppData\Roaming\Python\Python27\site-packages\octoprint_m33fio\__init__.py", line 4567, in on_event
    
  File "C:\Users\Neonode\AppData\Roaming\Python\Python27\site-packages\octoprint_m33fio\__init__.py", line 4427, in setFileLocations
    
  File "C:\Users\Neonode\AppData\Roaming\Python\Python27\site-packages\octoprint_m33fio\__init__.py", line 1195, in convertCuraToProfile
    # Remove temporary file
WindowsError: [Error 32] The process cannot access the file because it is being used by another process: 'c:\\users\\neonode\\appdata\\local\\temp\\tmpuwtd0o'
2016-07-24 21:48:07,880 - octoprint.plugins.softwareupdate - INFO - Saved version cache to disk
2016-07-24 21:49:23,180 - octoprint.server.util.sockjs - INFO - Client connection closed: 127.0.0.1
2016-07-24 21:49:24,145 - octoprint.server.util.sockjs - INFO - New connection from client: 127.0.0.1
2016-07-24 21:49:24,269 - octoprint.plugin - ERROR - Error while calling plugin m33fio
Traceback (most recent call last):
  File "C:\Users\Neonode\AppData\Roaming\Python\Python27\site-packages\octoprint-1.2.13-py2.7.egg\octoprint\plugin\__init__.py", line 204, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "C:\Users\Neonode\AppData\Roaming\Python\Python27\site-packages\octoprint_m33fio\__init__.py", line 4567, in on_event
    
  File "C:\Users\Neonode\AppData\Roaming\Python\Python27\site-packages\octoprint_m33fio\__init__.py", line 4427, in setFileLocations
    
  File "C:\Users\Neonode\AppData\Roaming\Python\Python27\site-packages\octoprint_m33fio\__init__.py", line 1195, in convertCuraToProfile
    # Remove temporary file
WindowsError: [Error 32] The process cannot access the file because it is being used by another process: 'c:\\users\\neonode\\appdata\\local\\temp\\tmptkwgfr'
2016-07-24 21:52:38,450 - octoprint.server.util.sockjs - INFO - Client connection closed: 127.0.0.1
2016-07-24 21:54:52,029 - octoprint.server.util.sockjs - INFO - New connection from client: 127.0.0.1
2016-07-24 21:54:52,204 - octoprint.plugin - ERROR - Error while calling plugin m33fio
Traceback (most recent call last):
  File "C:\Users\Neonode\AppData\Roaming\Python\Python27\site-packages\octoprint-1.2.13-py2.7.egg\octoprint\plugin\__init__.py", line 204, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "C:\Users\Neonode\AppData\Roaming\Python\Python27\site-packages\octoprint_m33fio\__init__.py", line 4567, in on_event
    
  File "C:\Users\Neonode\AppData\Roaming\Python\Python27\site-packages\octoprint_m33fio\__init__.py", line 4427, in setFileLocations
    
  File "C:\Users\Neonode\AppData\Roaming\Python\Python27\site-packages\octoprint_m33fio\__init__.py", line 1195, in convertCuraToProfile
    # Remove temporary file
WindowsError: [Error 32] The process cannot access the file because it is being used by another process: 'c:\\users\\neonode\\appdata\\local\\temp\\tmp5eme8f'

donovan6000, 2016-07-24:
M33 Fio V1.1 didn't work well on Windows, and it looks like it's also preventing you from updating for some reason. Run the following commands in the command prompt to update to the latest version of M33 Fio.
```
taskkill /IM pythonw.exe /T
ECHO y | "%SYSTEMDRIVE%\python\Scripts\pip.exe" uninstall OctoPrint-M33Fio
"%SYSTEMDRIVE%\python\Scripts\pip.exe" install https://github.com/donovan6000/M33-Fio/archive/master.zip --user
START "" "%SYSTEMDRIVE%\python\pythonw.exe" -c "import octoprint;octoprint.main()"
```