donovan6000/iMe #26: iMe and Simplify3d cause print to tilt

neonode, 2016-07-23:
Hello, 

I have issue with Simplify 3d directly connected to iMe firmware ver. 00.00.12. All my prints are tilted towards lower Y at 5 degrees. I have tested multiple backlash values but result is absolutely same  5 degree tilt towards lower Y. My printer works fine without tilt with stock firmware. It seems not to be backlash value as the form and the quality of the print is really good. Just this tilt. I am attaching pics.
[https://drive.google.com/open?id=0B_G0VYxtLPwGOV8xZmF4ZmQ3c1k](url)
[https://drive.google.com/open?id=0B_G0VYxtLPwGWWdIRHV2TmVSVzg](url)

Regards,
George

qwerty8224, 2016-07-23:
It seems to me that the case in the print speed is the same Mattecontrol

donovan6000, 2016-07-23:
It might be because the backlash speed is too high. Try reducing it and see if that has an affect. iMe moves the extruder faster than M3D's official firmware, so that could explain why your not experiencing the tilt when using M3D's official software.

I'll also take a look into the Y backlash compensation to see if it's not working correctly.

qwerty8224, 2016-07-23:
It does not help to decrease backlash speed

neonode, 2016-07-23:
Again same result with backlash speed 750. Same tilt 5 degree toward lower Y axis. Print quality is good again and the dimension of the circle is same across entire print. It seems that by some reason Y orientation change between every layer toward lower Y. I don't believe that this issue is hidden in backlash function as I saw it and seems correct and also according to the form I use to check the prints and the movement that printer do backlash error will cause deformation of the print which is not observed.

neonode, 2016-07-24:
Hi Donovan, 

I believe I catch what cause the tilt. Today I printed a cube 10x10x10mm. The cube is perfect (no tilt) except Y part is not 10mm but avarage 9.8mm and X is almost exactly 10.00mm. I examine the gcode for the cylinder and all curves are represented by lines with G1 instead controlled curves G2/G3. and because the steps per mm are not rounded number, when you calculate the distance you need to round the steps per this distance. Simplified 3D make circle as making clock and counted clock wise movement. Depending on the rounding this could cause an issue in my case. It is clear that the modifiers in the firmware for steps per mm differ for my printer as my Y slightly under gain equally in each layer and counting that you do equations for 10mm but I do 9.8mm this could cause slight shit that is independent from the backlash.

Could I ask you to add the ability in the firmware and in the M33 manager for changing steps per mm. I strongly believe that this will fix the issue and also will give the ability to the user to fine tune the printer for maximum accuracy.

Regards,
George

donovan6000, 2016-07-24:
Ok, I'll add adjusting the steps per mm in the next release. I have to work tomorrow, so it'll probably get released sometime on Monday.

Can you also try reducing the Y speed limit to something like 1000 and see if that helps at all. I'm still on the fence as to whether of not the issue is a result of the Y motor skipping at some speeds or the rounding that occurs when calculating the number of steps for the Y motor. My printer hasn't worked well for a few months, so I can't test things like this myself.

neonode, 2016-07-24:
Hi Donovan, 

I think I catch it. It comes from rounding. Rounding will work correct on linear movements but on curves they will cause spiral movement that will tilt the cylinder along the height. I tested your advice and it seems it do not affect the tilt angle at all. But I also changed slightly the dimension of the cylinder and surprise I got tilt for Y in opposite direction toward high Y and additionally tilt in X toward lower X axis. 

I sit and did some calculation and it seems that when you create cube it should not get tilt with handling the rounds but on curves you will get tilt as you have constant loose or gain during the build and you multiply the error within every layer. But overall the error within single layer is relatively small and it is equal in every layer so I can't see it within single layer dimension. 

I had same problem when I built my moto-focus controller for my telescope. So instead of rounding  the number of the steps I create a algorithm for handling the leftover. So in case I get 5.7 steps I launch only 5 and keep the rest in buffer, on next sequence I need to move 5.4 steps so I move this time 6 and keep leftover of 0.1 in the buffer. In your implementation with will be slightly more complex as you need to count the direction and to keep leftover from opposite moves with negative value but in general it should be easy for you compared to the effort you put here.  

Regards,
George

qwerty8224, 2016-07-25:
IME 010 M33Fio printed smoothly.

neonode, 2016-07-25:
Hi, from where I can download 010 in order to test.

qwerty8224, 2016-07-25:
I deleted at .... can u Donovan left

donovan6000, 2016-07-25:
All versions of iMe can be found [here](https://github.com/donovan6000/iMe/releases). V00.00.00.10 uses 32 micro steps per step, so the rounding error will built up much slowly over time when using it. The motors that the printer uses don't handle 32 micro steps per step very well which is why the movements are really jerky when using it. iMe V00.00.00.12 uses 8 micro steps per step which is what M3D's official firmware uses.

qwerty8224, 2016-07-25:
IME10 Mattercontrol problem, M33Fio no problem

neonode, 2016-07-25:
Qwerty8224, 

The problem is not in the Mattercontrol it is an co-incident of the speed that Mattercontrol use when creating the actual printer movement. With 32 micro-stepping if the motor can't handle it it will tend to go in resonance / miss steps on some speeds and will work correctly on other. So this is one of possible reasons why you have different result in different software. 

And I can confirm this motor used in this printer are the cheapest junk on the market ever. I have similar motors but quality built with gearbox and they cost me 50 euro per piece this ones used in Micro are 1.5 euro per piece.

neonode, 2016-07-25:
Donovan, 

This may be the issue why I want to get a rid of the stock firmware. They clear the error very strange. I believe they clear the leftover before changing the directing not on every move. This cause that I have never manage to get perfect circle. It is always comes deformed within single layer. I was on about to sell this shit when I saw iMe and I decide to give it last chance thanks to your effort.

qwerty8224, 2016-07-25:
Neonode You can view your engines?

neonode, 2016-07-25:
Okay I tested 0.10 ver. I can't extrude with it. It seems that 32 micro stepping do not give the desired torque.

neonode, 2016-07-25:
I didn't swap the stock crap. But for my other project I used this motor as I needed something precise and reliable: 
[http://www.mclennan.co.uk/product/p531-32-geared-stepper-motor](url) 

And this cost me 50 euro per pcs.

donovan6000, 2016-07-25:
[iMe's devel version](https://github.com/donovan6000/iMe/tree/devel) should be able to compensate for the rounding error now. Let me know if you guys still experience the tilt issue.

donovan6000, 2016-07-25:
Devel version also lets you set the steps per mm for each of the motors.

neonode, 2016-07-25:
Hi Donovan, 

New firmware don't work. Printer stop responding during first few commands. Just firmware hang and need restart. I can successfully move it with jog control, calibrate bed and etc. But when I execute code it stop between first and 10th line of G1 execution. It is something with function that looks like problem with the memory or buffer overflow as it do not sent back OK status to the software and you can't reconnect once this happen .

neonode, 2016-07-25:
Hi Donovan I am pasting you first few lines so you can check: 

;other temp:210
;ideal temp:210
G90
M106 S0
M104 S210 G90 ; Absolute positioning
G92 E0
G1 E-2.5000 F1800
G1 Z0.200 F100
G1 X37.189 Y34.992 F2160
G1 E0.0000 F540
G92 E0
G1 X38.511 Y33.591 E0.0721 F432
G1 X38.548 Y33.554 E0.0740
G1 X39.009 Y33.119 E0.0978
G1 X39.049 Y33.084 E0.0997
G1 X40.524 Y31.846 E0.1718
G1 X40.565 Y31.813 E0.1738
G1 X41.074 Y31.435 E0.1975
G1 X41.117 Y31.405 E0.1995
G1 X42.726 Y30.346 E0.2715
G1 X42.771 Y30.319 E0.2735
G1 X43.320 Y30.002 E0.2972
G1 X43.366 Y29.977 E0.2992
G1 X45.087 Y29.112 E0.3712
G1 X45.135 Y29.090 E0.3732
G1 X45.717 Y28.839 E0.3969
G1 X45.766 Y28.820 E0.3989
G1 X47.576 Y28.161 E0.4710
G1 X47.626 Y28.144 E0.4729
G1 X48.233 Y27.962 E0.4967
G1 X48.284 Y27.949 E0.4986
G1 X50.158 Y27.505 E0.5707
G1 X50.209 Y27.494 E0.5727
G1 X50.834 Y27.384 E0.5964
G1 X50.886 Y27.376 E0.5984
G1 X52.799 Y27.153 E0.6704
G1 X52.851 Y27.148 E0.6724
G1 X53.485 Y27.111 E0.6961
G1 X53.537 Y27.110 E0.6981
G1 X55.463 Y27.110 E0.7702
G1 X55.515 Y27.111 E0.7721
G1 X56.149 Y27.148 E0.7958
G1 X56.201 Y27.153 E0.7978
G1 X58.114 Y27.376 E0.8699
G1 X58.166 Y27.384 E0.8718
G1 X58.791 Y27.494 E0.8956
G1 X58.842 Y27.505 E0.8975
G1 X60.716 Y27.949 E0.9696
G1 X60.767 Y27.962 E0.9716
G1 X61.374 Y28.144 E0.9953
G1 X61.424 Y28.161 E0.9973
G1 X63.234 Y28.820 E1.0693
G1 X63.283 Y28.839 E1.0713
G1 X63.865 Y29.090 E1.0950
G1 X63.913 Y29.112 E1.0970
G1 X65.634 Y29.977 E1.1691
G1 X65.680 Y30.002 E1.1710
G1 X66.229 Y30.319 E1.1947
G1 X66.274 Y30.346 E1.1967
G1 X67.883 Y31.405 E1.2688
G1 X67.926 Y31.435 E1.2707
G1 X68.435 Y31.813 E1.2945
G1 X68.476 Y31.846 E1.2964
G1 X69.951 Y33.084 E1.3685
G1 X69.990 Y33.119 E1.3705
G1 X70.452 Y33.554 E1.3942
G1 X70.489 Y33.591 E1.3962
G1 X71.811 Y34.992 E1.4682
G1 X71.845 Y35.031 E1.4702
G1 X72.253 Y35.517 E1.4939
G1 X72.286 Y35.558 E1.4959
G1 X73.436 Y37.103 E1.5680
G1 X73.466 Y37.146 E1.5699
G1 X73.814 Y37.676 E1.5936
G1 X73.842 Y37.721 E1.5956
G1 X74.805 Y39.389 E1.6677
G1 X74.830 Y39.435 E1.6696
G1 X75.115 Y40.002 E1.6934
G1 X75.137 Y40.049 E1.6953
G1 X75.900 Y41.818 E1.7674
G1 X75.919 Y41.867 E1.7694
G1 X76.136 Y42.463 E1.7931
G1 X76.153 Y42.513 E1.7951
G1 X76.705 Y44.358 E1.8671
G1 X76.719 Y44.408 E1.8691
G1 X76.865 Y45.026 E1.8928
G1 X76.875 Y45.077 E1.8948
G1 X77.210 Y46.974 E1.9669

qwerty8224, 2016-07-25:
Printing is exactly but disconnected extruder E + E

donovan6000, 2016-07-26:
Try out the latest devel version and let me know if that fixes the issue where the printer becomes unresponsive.

neonode, 2016-07-26:
It is much more stable but again after a 2-3 mins it hang. It hang so hard that it stop to control the heater.

donovan6000, 2016-07-26:
Ok, it seems like this is being caused by a bug in the compiler, assembler, or linker. Try out the latest devel versions again and let me know if it's any better.

neonode, 2016-07-26:
Same. Can't survive the base.

qwerty8224, 2016-07-26:
Mattercontrol prints for 20 minutes

neonode, 2016-07-26:
My mistake. It works now just I re flash with wrong 00.13. The results will be here soon.

neonode, 2016-07-26:
Donovan, I am really clueless now. I have never saw thing like this. I created two prints with different backlash values and you can see the result. Print is absolutely same nothing have been change in Simplify3d. Backlash value should generate only static error without multiplication but you can see the result. Also Backlash value can cause tilt only in really drastic situations. This is either from missing steps + error from the function or I don't know really this is something very very strange.

[https://drive.google.com/open?id=0B_G0VYxtLPwGMWpjOVNmOEVVTlU](url)
[https://drive.google.com/open?id=0B_G0VYxtLPwGN3dQdUdIMTVSMVk](url)
[https://drive.google.com/open?id=0B_G0VYxtLPwGVUQ4dVBiTW5tdWc](url)

donovan6000, 2016-07-26:
Wow, that's pretty weird. Lets see if it's actually rounding error that's causing the tilt to begin with. The newest [devel version](https://github.com/donovan6000/iMe/tree/devel), V00.00.01.00, has the fractional step compensation removed, so use M33 Manager to set the steps per mm for each of the motors to whole numbers and print something so that we can see if the rounding error is really the source of the problem.

Also thanks guys for testing all this stuff :)

qwerty8224, 2016-07-26:
Steps 18 and 19 additions Y have

neonode, 2016-07-26:
Hi Donovan, 

I just fail back to 00.12 to check what going on and if my printer work correctly. As I expect it work. I print out same detail and I got previous result straight tilt all along the axis not curved as in 00.13. I also changed my backlash to BX:0 and BY:0. As It should be it didn't affect the tilt at all just deform the circle within the layer and I didn't manage to close the loop. I can send a pics if you want. 

Than I went in to my workshop as I got CNC to see how my machine deal with this rounds. My stepper motors use very weird ratio 533.33333 steps per 1mm. And I was surprised. My CNC mill will never go on a position that it not round number of steps. If I call a position that is not  round number divided by the ratio my CNC will go to the nearest possible location and will update the grid position not with the requested but with the actual one. So you may think to implement this. it should be fairly easy according to your code and will reduce the overhead of keeping and dealing with leftovers and rounds. I wish I could help you but I am not good with Atmel studio at all neither with the Technics that you use ( timers/counters and interrupts). You are far more advance than me so I prefer not to interact with code.

qwerty8224, 2016-07-27:
step 18 19 octoprint smooth 
I tested on

donovan6000, 2016-07-27:
Ok, so it turns out that there can still be a fractional number of steps even when the steps per mm are whole numbers if the distance traveled has a certain fractional component. So MatterControl's slicer must have had more movements with certain fractional components than OctoPrint which explains the bias occurring in MatterControl but not OctoPrint.

Try out the latest [devel version](https://github.com/donovan6000/iMe/tree/devel), V00.00.01.01, and let me know if the tilt is fixed. If it's still not fixed then I'll look into implementing neonode's actual grid position idea.

qwerty8224, 2016-07-27:
While prints exactly Mattercontrol

donovan6000, 2016-07-27:
Try [devel version](https://github.com/donovan6000/iMe/tree/devel) V00.00.01.02

qwerty8224, 2016-07-27:
))) I'm at work.

qwerty8224, 2016-07-27:
displacement has X Y IME 1.02

neonode, 2016-07-28:
Hi Donovan,

Yes I confirm. Everything is fixed prints are awesome now. At the end I can print circle thanks to you and your great work!

Please give more information what caused the issue and how you fixed it.

qwerty8224, 2016-07-28:
Thank you ..... I get home check otpishus

qwerty8224, 2016-07-28:
When RELEASE ? )))))))

qwerty8224, 2016-07-28:
Other printing
exactly knocked layer 4 and then again exactly (AND)

donovan6000, 2016-07-28:
The problem was a result of the rounding errors when calculating the number of steps, so neonode's was right about that. When I was initially trying to fix it, I didn't notice that I was accounting for the remaining fractional number of steps twice due to how iMe implements the bed leveling. That caused the [double tilt](https://drive.google.com/open?id=0B_G0VYxtLPwGN3dQdUdIMTVSMVk). I set it up so that I could simulate all the movements on my computer yesterday which allow me to see how the number of steps changes over time, and I was able to identify the problem.

The second problem where iMe would crash after a few seconds/minutes was due to the stack overflowing into the data/bss in RAM. So that wasn't a compiler error. Trying to fit all the backlash and bed compensation into iMe while keeping it less than the size of the microcontroller's flash size, 32K, hasn't been easy, and a lot of the optimizations I've done has reduced the firmware's size at the cost of increasing the data/bss. To fix this I removed the support for host commands in the G-code since they're not used by anything, which freed up a pretty big chuck of data/bss.

I'll probably do a release sometime today.

neonode, 2016-07-28:
Hi Donovan,

Thanks for the clarification.

closed 2016-07-28 by neonode