donovan6000/iMe #27: gantry movement confusion

analogmonster, 2016-07-23:
not happening every print, but I have regularly been having an issue.  When it homes at the beginning of the print it does x correctly but then instead of homing y it drives x hard to the left with grinding (ie accelerometer seems to be ignored) so I abort the print.  However I have also successfully homed then printed fine, and then the next test border fails again.  

I'm currently just printing the test border each time so there definately isnt an issue with the model.  I reorinted the bed and did a full bed recalibration.  Switing back to stock for now but please let me know if you have any idea what is happening, thanks

neonode, 2016-07-23:
I use v00.00.12 and I use g28/30 to home and it works fine for me. G32 also work fine.

analogmonster, 2016-07-24:
Thanks for replying. Yeah that's why I just switched back to stock, it clearly wasn't something everyone was suffering from. I'm using octoprint so I will just stick with optimising cura profiles using stock unless someone can explain what's going on

donovan6000, 2016-07-24:
It sounds like the Y jerk is too sensitive, so it's detecting that it has hit the right side immediately after hitting the back wall. We'll decrease the [sensitivity](https://github.com/donovan6000/iMe/blob/master/accelerometer.h#L7-L8) a bit and/or add a delay before homing Y in the next version of iMe which should help.

The accelerometer that the printer uses isn't very good, so it's difficult to determine when the extruder has hit the walls when homing. That's why M3D's official firmware doesn't use the acceleromoeter when homing.

analogmonster, 2016-07-28:
Thanks, that was pretty much what I had assumed was going on. I look forward to the next update

closed 2016-07-28 by analogmonster

donovan6000, 2016-07-29:
Just release V00.00.01.06 which has much less sensitivity when Y homing. Hopefully that helps.