donovan6000/iMe #28: Calibration bed pozition error Z0 Y-axis

qwerty8224, 2016-07-28:


donovan6000, 2016-07-28:
Try out version [V00.00.01.05 ](https://github.com/donovan6000/iMe/tree/devel) and let me know if the homing is still too sensitive.

Also let me know if the calibrate bed center Z0, G30, command is working well. It was calibrating it slightly too low on my printer, so I have it set [Z0 to 0.2mm higher](https://github.com/donovan6000/iMe/blob/devel/motors.cpp#L22) than what it would normally.

qwerty8224, 2016-07-28:
The problem comes with time

donovan6000, 2016-07-28:
What do you mean?

qwerty8224, 2016-07-28:
2 Calibration of the 10 cases

donovan6000, 2016-07-29:
Is it just the Y (back edge) that is too sensitivity, or do you get it on the X (right edge) sometimes too?

qwerty8224, 2016-07-29:
Not only Y

donovan6000, 2016-07-29:
Ok, try the newest [devel version](https://github.com/donovan6000/iMe/tree/devel). It still has the same version number, V00.00.01.05, since I didn't feel like changing it.

I wonder if I should make the X and Y sensitivities adjustable. That devel version isn't sensitive enough for my printer, so the Y homing just grinds into the back edge for a while like M3D's official firmware. So all printer must be slightly different due to wear.

qwerty8224, 2016-07-29:
Can play the extruder axis Y and blame

qwerty8224, 2016-07-29:
Reducing the sensitivity and correct the problem
https://drive.google.com/file/d/1vEXnyD6A8J4le2eJqqKnz9RIIW819Kh3yA/view?usp=sharing

donovan6000, 2016-07-29:
Those prints turned out well :)

The speed limits get applied to the backlash speed, so it was actually getting changed from 2700 to 1500 if your XY speed limits were 1500. Each motor will never go faster than its speed limit, so if you want to use a backlash speed of 2700 then you need to increase your XY speed limits to >= 2700.

I didn't change anything about the sensing used in G30, so it was just a coincidence that it's sensing more in V00.00.01.06. The acceleromoter doesn't detect Z movement very well since the Z motor moves so slowly, so G30 senses when the extruder housing tilts due to the nozzle hitting the bed. It needs to be really sensitive to detect right when that happens, so any kind of vibrations can cause it to falsely sense.

qwerty8224, 2016-07-29:
Thank you ....
The main thing that all the prints exactly)))))

closed 2016-07-29 by qwerty8224