donovan6000/iMe #30: Occasional calibration issues

analogmonster, 2016-08-01:
00.00.01.06 got rid of my intermittent y axis freak out when homing before printing. If anything y is not sensitive enough now. As you previously mentioned in another topic, adjusting accelerometer thresholds manually would be useful (for me at least). 

However there was weird behaviour when calibrating centre zero and bed orientation.

When I first uploaded the ime firmware and octoprint says recalibrate, I cancelled and then did it manually (I make a habit of power cycling everything after a firmware upgrade) It decided z0 centre was about 20mm away from the bed. Set it manually to very near the bed, then auto, same issue again. Then when it did orientation I even had to remove the power cable, because it was going z+ until the top of the extruder enclosure hit the frame when doing rear right calibration.

Go to stock, calibrate as normal. Switch to ime, repeat above exactly and same issue.

However, I have it calibrating fine provided I don't cancel the first z calibration prompt that comes automatically after installing the ime firmware. Z doesn't go too + before heading - and zero is found actually at the bed.

Out of interest, do you think it would be possible to upgrade the accelerometer? Presumably if I design a pcb the same dimensions, with the same connector but a better accelerometer that uses the same protocol (spi/i2c/analog, will look later) it would be a slot in replacement

qwerty8224, 2016-08-01:
Confirm calibration in MetterControl and Oktoprint not match.

donovan6000, 2016-08-03:
That's kinda weird since the calibration it does for the '[Invalid bed plane](https://github.com/donovan6000/M33-Fio/blob/master/octoprint_m33fio/static/js/m33fio.js#L12275-L12282)' message is identical to the calibration used by the '[Calibrate bed center Z0](https://github.com/donovan6000/M33-Fio/blob/master/octoprint_m33fio/static/js/m33fio.js#L8631-L8638)' button, which is the G30 command, so both of those should work exactly the same.

The G30 procedure is very sensitive to any kind of external vibrations, so something like a gust of wind could make it falsely detect Z0. Or the accelerometer could have malfunctioned during the procedure at some point. iMe only checks that the accelerometer is working when it starts up, so something like that could happen.

I'll add a limit to prevent it from moving up higher than the initial Z when starting the calibration to prevent it from going too far up. I'll also add detection for if the accelerometer malfunctions. Hopefully doing those things will fix this issues your having.

I don't think it's possible to do a slot in replacement for the accelerometer without modifying the firmware as well. Both iMe and M3D's official firmware verify that the acceleromoeter is working by requesting its [ID](https://github.com/donovan6000/iMe/blob/master/accelerometer.cpp#L26). So a replacement accelerometer would have to have the same ID and support the same commands. The accelerometer they use is the [MMA8652FC](http://cache.nxp.com/files/sensors/doc/data_sheet/MMA8652FC.pdf), and here's the pin out for the board's header if your still interested.
[image]
Pin1 Purple VCC 3.3V
Pin2 Black GND
Pin3 Black SDA
Pin4 Brown Enable
Pin5 Brown SCL

analogmonster, 2016-08-03:
That's great thanks, saved me from opening my m3d up again.

I've never been sure the accelerometer is working correctly.  It works a lot of the time but I have had it miss in calibration and end up pushing the x axis rods off the y axis completely.  Then when I swapped to ime I have other accellerometer issues.  I dont want to have to send it back, its very expensive for what they do.

So were looking at an I2C 3 axis 12 bit accelerometer.  It has ±2.5% accuracy and ±0.0074%/*C temperature sensitivity as well as a noise level of 150 micro g per root Hertz.

I looked at NXPs range, there isnt much that's better. chipan issue

analogmonster, 2016-08-05:
hmmm.  Well I've read countless datasheets over the past few days.  There aren't many 12 bit i2c accelerometers that offer much of an improvement - perhaps a little less self noise but barely an improvement of accuracy and temperature stability.

So I looked at 14 and 16 bit.  Again, no significant improvement beyond resolution however that could make all the difference.  But implementing 16 bit would increase the time it took for passing data out of accelerometer to the microcontroller, as well as an increase in the time the microcontroller took to do the sample averaging, which may mean compromising on sample number.

In conclusion I give up up this idea.  It's a lot of work for possibly not much improvement.  Were I designing from scratch it would be a totally different situation (eg choose a microcontroller with more SPI channels.  Use SPI for accelerometer since its WAY faster, use 3 different accelerometers since there's always one axis with less accuracy and use interrupts for collision detection).  I will have a think about implementing it that way when I build my own printer later in the year.  

And since I didn't make it clear earlier I place a large cardboard box around the printer (at least until Chase delivers my heated bed and enclosure) , keep the door and window of the room shut and I'm usually in a different room entirely while its printing.  It's on a stable level surface.  I've a stable physical environment but I haven't checked if my mains voltage is misbehaving - I have no reason to suspect it is though.  I look forward to seeing your next firmware release, thanks

closed 2016-08-05 by analogmonster

donovan6000, 2016-08-05:
I just added most of the stuff we talked about to [iMe's devel version](https://github.com/donovan6000/iMe/tree/devel), like constantly checking if the accelerometer is working and preventing it from going higher than the starting height when calibrating Z0.

It does reads the X and Y jerk sensitivity for homing from the printer's EEPROM, but M33 Manager doesn't allow setting them yet. So the only way to adjust them right now is with M33 Fio's EEPROM writer. Address 0x29D is the X sensitivity and 0x29E is the Y sensitivity. 0 is the lowest sensitivity and 255 is the highest sensitivity.

I think a lot of modern 3D printers use [endstops](http://reprap.org/wiki/Mechanical_Endstop) for homing and inductive proximity sensors for bed leveling instead of an accelerometer, so you might want to look into those when you build a 3D printer.

analogmonster, 2016-08-05:
Great, hopefully I will have time to look today after the current print finishes. 

I have experience with end stops, added them to my Chinese cnc milling machine. And I've an inductive proximity sensor I got recently to see how well it would work with the pick and place machine I'm working on at the moment. 

I'm interested in adding accelerometers near the print head for my own machine, not so much for calibration/homing but fault detection. I'd far prefer hardware calibration to software any day