donovan6000/iMe #33: Offset Y axis X

qwerty8224, 2016-08-03:
Through time. Tried different speed and backslash

donovan6000, 2016-08-03:
The motors will skip and cause shifts like that if they move too fast because of the printer's mechanical design. This isn't something that can be fixed in iMe since it's hardware related, so you'll have to keep the speed limits and backlash speed slow enough to prevent the motors from skipping.

qwerty8224, 2016-08-03:
The same file prints M3D firmware .... but at a slower rate.
IME increases the speed?

donovan6000, 2016-08-03:
Yes, iMe prints faster than M3D's official firmware since it's able to get closer to a commands specified feedrate than M3D's official firmware does.

The firmware has to implement delays to enforce the speed limits and to make sure that all motors end at the same time for each command, and the implementation of these delays can cause a motor to move slower than the commands provided feedrate. For example, if the command G0 X10 F1000 is being run then the actual movement speed when using iMe will be closer to F1000 than when using M3D's official firmware. That means iMe's movement speeds are more accurate when compared to the file being printed, and that causes it to print faster than M3D's official firmware.

qwerty8224, 2016-08-03:
Thank you. We will seek speed)

closed 2016-08-04 by qwerty8224