donovan6000/iMe #36: Octoprint 1.6 IME 1.12

qwerty8224, 2016-08-19:
Movement is to big /// 
Settings 1500mm / s XY
infill 16 
inside 14
outside 12
travel 20

donovan6000, 2016-08-20:
Can you upload the G-code file that your trying to print? The 'Movement is too big' error should only happen when an axis attempts to move more than 4294967295 steps, which shouldn't ever happen for a printer this small.

Here's what 4294967295 steps means in terms of millimeters for each axis if using the default steps/mm.
4294967295 steps = 27807366mm for the X axis
4294967295 steps = 29811504mm for the Y axis
4294967295 steps = 830645mm for the Z axis
4294967295 steps = 4179565mm for the E axis

qwerty8224, 2016-08-22:
The problem occurred when connecting the printer without file gcode
It seems to me that the flashing takes place is not always successful. Maybe there are traces or make erasing before flashing?

qwerty8224, 2016-08-22:
Now 3 times to print all good ....)))

qwerty8224, 2016-08-23:
;Generated from feeder_v6_arm.STL 80b3347636df3edc57157f37c2063f06aa507e94
;Generated with Cura_SteamEngine 15.04.6
M109 T0 S215.0
T0
; Sliced at: Tue 23-08-2016 03:12:53
; Basic settings: Layer height: 0.2 Walls: 1.4 Fill: 8.75

;Layer count: 70
;LAYER:0
M107
G0 F1200 X39.515 Y33.345 Z0.300
;TYPE:SKIRT
G1 F600 X40.806 Y31.857 E0.08600
G1 X41.628 Y30.976 E0.13860
G1 X62.078 Y30.976 E1.03132
G1 X62.582 Y30.995 E1.05334
G1 X63.181 Y31.066 E1.07967
G1 X64.133 Y31.294 E1.12240
G1 X64.761 Y31.523 E1.15158
G1 X65.642 Y31.970 E1.19471
G1 X66.161 Y32.314 E1.22189
G1 X66.633 Y32.688 E1.24818
G1 X67.328 Y33.380 E1.29099
G1 X67.713 Y33.865 E1.31803
G1 X68.051 Y34.369 E1.34452
G1 X68.495 Y35.241 E1.38723
G1 X68.714 Y35.824 E1.41442
G1 X68.877 Y36.409 E1.44093
G1 X69.037 Y37.402 E1.48484
G1 X69.059 Y38.437 E1.53003
G1 X68.988 Y39.503 E1.57667
G1 X68.872 Y40.195 E1.60730
G1 X68.584 Y41.246 E1.65487
G1 X68.328 Y41.904 E1.68569
G1 X67.836 Y42.869 E1.73298
G1 X67.480 Y43.425 E1.76180
G1 X67.121 Y43.906 E1.78800
G1 X66.732 Y44.361 E1.81413
G1 X66.314 Y44.786 E1.84015
G1 X65.839 Y45.210 E1.86795
G1 X64.961 Y45.846 E1.91528
G1 X64.354 Y46.201 E1.94597
G1 X63.364 Y46.648 E1.99339
G1 X62.698 Y46.875 E2.02411
G1 X61.638 Y47.114 E2.07154
G1 X60.982 Y47.197 E2.10041
G1 X57.860 Y47.358 E2.23688
G1 X57.844 Y46.569 E2.27133
G1 X57.592 Y46.663 E2.28307
G1 X57.049 Y46.914 E2.30918
G1 X56.551 Y47.191 E2.33406
G1 X56.396 Y47.294 E2.34218
G1 X56.396 Y58.725 E2.84119
G1 X56.929 Y59.235 E2.87339
G1 X56.912 Y70.573 E3.36834
G1 X56.912 Y82.760 E3.90035
G1 X56.730 Y83.636 E3.93941
G1 X56.196 Y84.479 E3.98297
G1 X55.677 Y84.953 E4.01365
G1 X55.063 Y85.299 E4.04442
G1 X54.456 Y85.523 E4.07267
G1 X53.694 Y85.621 E4.10620
G1 X46.052 Y85.620 E4.43981
G1 X42.249 Y85.621 E4.60582
G1 X41.716 Y85.571 E4.62919
G1 X40.979 Y85.388 E4.66234
G1 X40.084 Y84.881 E4.70725
G1 X39.408 Y84.117 E4.75178
G1 X39.065 Y83.327 E4.78938
G1 X38.878 Y82.308 E4.83460
[... 265 lines cut ...]
G1 X56.493 Y36.581 E22.49947
G1 X56.355 Y37.138 E22.52452
G1 X56.275 Y37.704 E22.54947
G1 X56.253 Y38.276 E22.57446
G1 X56.291 Y38.849 E22.59953
G1 X56.387 Y39.414 E22.62454
G1 X56.540 Y39.966 E22.64955
G1 X56.749 Y40.500 E22.67458
G1 X57.012 Y41.009 E22.69959
G1 X57.325 Y41.489 E22.72461
G1 X57.687 Y41.934 E22.74965
G1 X58.091 Y42.339 E22.77462
G1 X58.256 Y42.476 E22.78399
G1 X58.250 Y42.831 E22.79948
G1 X58.253 Y43.035 E22.80839
G1 X58.231 Y43.243 E22.81752
G1 X58.178 Y43.434 E22.82617
G1 X58.092 Y43.589 E22.83391
G1 X57.995 Y43.677 E22.83963
G1 X57.879 Y43.735 E22.84529
G1 X57.288 Y43.895 E22.87202
G1 X56.547 Y44.171 E22.90654
G1 X55.830 Y44.503 E22.94103
G1 X55.139 Y44.887 E22.97554
G1

donovan6000, 2016-08-23:
Do you only get the error when your connecting to the printer before you even send any commands to it? Or do you get the error every time you try printing that G-code file?

Can you enable OctoPrint's serial logging and upload the octoprint.log file after you get the error?
[image]
[image]

qwerty8224, 2016-08-23:
When I connect and when print error occurs.
I noticed that the error occurs in the transition from a Mattercontrol->  M33fio

qwerty8224, 2016-08-24:
2016-08-24 19:19:05,917 Original Response: PROTOCOL:RepRap FIRMWARE_NAME:iMe FIRMWARE_VERSION:00.00.01.15 MACHINE_TYPE:Micro_3D SERIAL_NUMBER:OR15081101050016

2016-08-24 19:19:05,918 Processed Response: PROTOCOL:RepRap FIRMWARE_NAME:iMe FIRMWARE_VERSION:00.00.01.15 MACHINE_TYPE:Micro_3D SERIAL_NUMBER:OR15081101050016

2016-08-24 19:19:05,918 Original Sent: M117

2016-08-24 19:19:05,920 Processed Sent: M117
2016-08-24 19:19:05,921 Original Sent: M114

2016-08-24 19:19:05,921 Processed Sent: M114
2016-08-24 19:19:05,921 Original Response: ok

2016-08-24 19:19:05,921 Processed Response: ok

2016-08-24 19:19:05,921 Original Sent: M619 S12 T4

2016-08-24 19:19:05,923 Processed Sent: M619 T4 S12
2016-08-24 19:19:05,923 Original Response: XV:1 YV:1 ZV:1

2016-08-24 19:19:05,923 Processed Response: XV:1 YV:1 ZV:1

2016-08-24 19:19:05,924 Original Response: ok

2016-08-24 19:19:05,924 Processed Response: ok

2016-08-24 19:19:05,924 Original Sent: M619 S16 T4

2016-08-24 19:19:05,924 Processed Sent: M619 T4 S16
2016-08-24 19:19:05,924 Original Response: X:0.0000 Y:-0.0000 Z:-1476394240.0000 E:0.0000

2016-08-24 19:19:05,926 Processed Response: X:0.0000 Y:-0.0000 Z:-1476394240.0000 E:0.0000

2016-08-24 19:19:05,926 Original Response: ok

2016-08-24 19:19:05,926 Processed Response: ok

2016-08-24 19:19:05,926 Original Sent: M619 S20 T4

2016-08-24 19:19:05,927 Processed Sent: M619 T4 S20
2016-08-24 19:19:05,927 Original Response: PT:12 DT:1051595899

2016-08-24 19:19:05,927 Processed Response: PT:12 DT:1051595899

2016-08-24 19:19:05,927 Original Response: ok

2016-08-24 19:19:05,927 Processed Response: ok

2016-08-24 19:19:05,927 Original Sent: M619 S24 T4

2016-08-24 19:19:05,927 Processed Sent: M619 T4 S24
2016-08-24 19:19:05,927 Original Response: PT:16 DT:1070134723

2016-08-24 19:19:05,928 Processed Response: PT:16 DT:1070134723

2016-08-24 19:19:05,928 Original Response: ok

2016-08-24 19:19:05,928 Processed Response: ok

2016-08-24 19:19:05,928 Original Sent: M619 S28 T4

2016-08-24 19:19:05,930 Processed Sent: M619 T4 S28
2016-08-24 19:19:05,930 Original Response: PT:20 DT:0

2016-08-24 19:19:05,930 Processed Response: PT:20 DT:0

2016-08-24 19:19:05,930 Original Response: ok

2016-08-24 19:19:05,930 Processed Response: ok

2016-08-24 19:19:05,930 Original Sent: M619 S32 T4
[... 329 lines cut ...]
2016-08-24 19:19:14,581 Processed Response: 
2016-08-24 19:19:15,604 Original Response: wait

2016-08-24 19:19:15,604 Processed Response: 
2016-08-24 19:19:15,605 Original Response: rs 2

2016-08-24 19:19:15,605 Processed Response: 
2016-08-24 19:19:16,608 Original Response: wait

2016-08-24 19:19:16,608 Processed Response: 
2016-08-24 19:19:16,608 Original Response: rs 2

2016-08-24 19:19:16,608 Processed Response: 
2016-08-24 19:19:17,614 Original Response: wait

2016-08-24 19:19:17,614 Processed Response: 
2016-08-24 19:19:17,614 Original Response: rs 2

2016-08-24 19:19:17,614 Processed Response: 
2016-08-24 19:19:18,614 Original Response: wait

2016-08-24 19:19:18,614 Processed Response: 
2016-08-24 19:19:18,628 Original Response: rs 2

2016-08-24 19:19:18,628 Processed Response:

qwerty8224, 2016-08-24:
2016-08-24 19:25:34,700 Original Response: PROTOCOL:RepRap FIRMWARE_NAME:iMe FIRMWARE_VERSION:00.00.01.15 MACHINE_TYPE:Micro_3D SERIAL_NUMBER:OR15081101050016

2016-08-24 19:25:34,700 Processed Response: PROTOCOL:RepRap FIRMWARE_NAME:iMe FIRMWARE_VERSION:00.00.01.15 MACHINE_TYPE:Micro_3D SERIAL_NUMBER:OR15081101050016

2016-08-24 19:25:34,700 Original Sent: M117

2016-08-24 19:25:34,700 Processed Sent: M117
2016-08-24 19:25:34,701 Original Sent: M114

2016-08-24 19:25:34,701 Processed Sent: M114
2016-08-24 19:25:34,703 Original Response: ok

2016-08-24 19:25:34,703 Processed Response: ok

2016-08-24 19:25:34,703 Original Sent: M619 S12 T4

2016-08-24 19:25:34,703 Processed Sent: M619 T4 S12
2016-08-24 19:25:34,703 Original Response: XV:1 YV:1 ZV:1

2016-08-24 19:25:34,703 Processed Response: XV:1 YV:1 ZV:1

2016-08-24 19:25:34,703 Original Response: ok

2016-08-24 19:25:34,703 Processed Response: ok

2016-08-24 19:25:34,704 Original Sent: M619 S16 T4

2016-08-24 19:25:34,704 Processed Sent: M619 T4 S16
2016-08-24 19:25:34,706 Original Response: X:0.0000 Y:-0.0000 Z:-1476394240.0000 E:0.0000

2016-08-24 19:25:34,706 Processed Response: X:0.0000 Y:-0.0000 Z:-1476394240.0000 E:0.0000

2016-08-24 19:25:34,706 Original Response: ok

2016-08-24 19:25:34,706 Processed Response: ok

2016-08-24 19:25:34,707 Original Sent: M619 S20 T4

2016-08-24 19:25:34,707 Processed Sent: M619 T4 S20
2016-08-24 19:25:34,707 Original Response: PT:12 DT:1051595899

2016-08-24 19:25:34,707 Processed Response: PT:12 DT:1051595899

2016-08-24 19:25:34,707 Original Response: ok

2016-08-24 19:25:34,707 Processed Response: ok

2016-08-24 19:25:34,707 Original Sent: M619 S24 T4

2016-08-24 19:25:34,709 Processed Sent: M619 T4 S24
2016-08-24 19:25:34,709 Original Response: PT:16 DT:1070134723

2016-08-24 19:25:34,709 Processed Response: PT:16 DT:1070134723

2016-08-24 19:25:34,709 Original Response: ok

2016-08-24 19:25:34,710 Processed Response: ok

2016-08-24 19:25:34,710 Original Sent: M619 S28 T4

2016-08-24 19:25:34,710 Processed Sent: M619 T4 S28
2016-08-24 19:25:34,710 Original Response: PT:20 DT:0

2016-08-24 19:25:34,710 Processed Response: PT:20 DT:0

2016-08-24 19:25:34,710 Original Response: ok

2016-08-24 19:25:34,710 Processed Response: ok

2016-08-24 19:25:34,710 Original Sent: M619 S32 T4
[... 383 lines cut ...]
2016-08-24 19:25:51,818 Processed Response: 
2016-08-24 19:25:52,836 Original Response: wait

2016-08-24 19:25:52,836 Processed Response: 
2016-08-24 19:25:52,838 Original Response: rs 2

2016-08-24 19:25:52,838 Processed Response: 
2016-08-24 19:25:53,838 Original Response: wait

2016-08-24 19:25:53,838 Processed Response: 
2016-08-24 19:25:53,838 Original Response: rs 2

2016-08-24 19:25:53,838 Processed Response: 
2016-08-24 19:25:54,842 Original Response: wait

2016-08-24 19:25:54,842 Processed Response: 
2016-08-24 19:25:54,842 Original Response: rs 2

2016-08-24 19:25:54,842 Processed Response: 
2016-08-24 19:25:55,842 Original Response: wait

2016-08-24 19:25:55,842 Processed Response: 
2016-08-24 19:25:55,842 Original Response: rs 2

2016-08-24 19:25:55,842 Processed Response:

donovan6000, 2016-08-24:
2016-08-24 19:25:34,701 Original Sent: M114
2016-08-24 19:25:34,706 Original Response: X:0.0000 Y:-0.0000 Z:-1476394240.0000 E:0.0000

Your printer thinks that its Z position is at -1476394240, and that's causing the the problem.

The Z position can get corrupted like that if you use M3D's official software to install M3D's firmware onto a printer that currently has iMe firmware since the firmwares store the Z position differently in the printer's EEPROM. M33 Fio and M33 Manager automatically convert the Z position in EEPROM when switching between the different firmwares so that the Z position doesn't get corrupted.

But if your going directly from MatterControl to OctoPrint then your not changing the printer's firmware, so the Z position shouldn't be getting corrupted. The Z position can also get corrupted if you remove the printer's power exactly when the Z value in the printer's EEPROM is being updated.

Can you run an M114 command in MatterControl before you connect the printer to OctoPrint? That will let us know if MatterControl is to blame.

qwerty8224, 2016-08-24:
The problem arises without Mattecontrol

closed 2016-08-24 by qwerty8224