donovan6000/iMe #37: Question?

qwerty8224, 2016-08-19:
You can add a change CALIBRATE_Z0_CORRECTION?

donovan6000, 2016-08-20:
I'll add it into the next version of iMe.

qwerty8224, 2016-08-22:
Thanks I think many like it.

closed 2016-08-22 by qwerty8224

donovan6000, 2016-08-23:
Newest [devel version](https://github.com/donovan6000/iMe/tree/devel) allows setting the calibrate Z0 correction. Positive values mean higher up, and negative values mean lower down. So a calibrate Z0 correction of +2.7mm means to move the Z0 up by 2.7mm higher than it sensed it to be.

dieideeistgut, 2016-09-08:
Hello - sorry to re-open. Found that you can't set the Z0 correction to a negative number because "min" html-attribute on the field itself is set to a positive number. Removing this (via Devtools, or such) lets me set negative values though. Just wanted to drop this note and leave a 👍 for all the work that has been done here.

donovan6000, 2016-09-11:
@dieideeistgut Thanks. It's fixed in [M33 Fio's devel version](https://github.com/donovan6000/M33-Fio/tree/devel) now.