donovan6000/iMe #40: iME 1.13 Bias backslash 900 IME 1.12 backslash 1500 GOOD

qwerty8224, 2016-08-23:


donovan6000, 2016-08-23:
I improved the software delay algorithm in iMe 1.13 and 1.14 to make the speed as fast as possible. Before it would move at up to 30% slower than the specified feed rate due to how the delays were calculated, but now it moves at up to 1% slower than the specified feed rate.

iMe 1.12, G0 X10 F1500 would move at 1500mm/min - 30% = 1050mm/min worst case
iMe 1.13, G0 X10 F1500 would move at 1500mm/min - 1% = 1485mm/min worst case

So the offset might actually be cause by the the XY speed limits being too high instead of the backlash speed being too high. So see if reducing your XY speed limits helps. If it doesn't fix it then I'll use the old delay algorithm to make the movements slower again.

qwerty8224, 2016-08-23:
I noticed the printer began to fly, and print crooked))))
Do not targeted it at such speed. A reduction in delay is very good !!!
And from the plastic, I realized how much depends here now I have a poor quality plastic, too.

closed 2016-08-24 by qwerty8224