donovan6000/iMe #42: Z is broken

qwerty8224, 2016-08-27:
The value Z breaks when switching from M3D manager at MatterControl

qwerty8224, 2016-08-28:
Once!!! XYZ values transfer correctly

donovan6000, 2016-08-31:
I tried about 20 times to recreate this issue, but so far I've haven't had any problems. None of my X, Y, or Z values are corrupted when switching between M33 Manager and MatterControl.
[image]

Can you post everything in MatterControl's terminal output when you experience this issue?

qwerty8224, 2016-08-31:
Ok how to get problemmu lay.
Ultimaker I finish a project and do not touch the printer .... until it works well)))

qwerty8224, 2016-08-31:
From Mattecointrol a M3D manager
[image]

qwerty8224, 2016-09-18:
IME 1.20 problem (

donovan6000, 2016-09-19:
I still haven't been able to recreate this issue. Does it happen if you try it on another computer?

qwerty8224, 2016-09-19:
Strange .... What you do not have a problem !!!! Maybe I do not have ....
I check the house again from scratch.

donovan6000, 2016-09-21:
Do you still have this problem when using an [older version of M33 Manager](https://github.com/donovan6000/iMe/releases)?

donovan6000, 2016-09-21:
This problems happens for you when switching between firmware mode and bootloder mode using only M33 Manager?

qwerty8224, 2016-09-21:
Yes

donovan6000, 2016-09-21:
Does the problem happen with older versions of M33 Manager?

qwerty8224, 2016-09-21:
Yes v0.10 v0.12 problemm

donovan6000, 2016-09-21:
Does it happen with the first version of M33 Manager [V0.1](https://github.com/donovan6000/iMe/archive/V00.00.00.06.zip)?

qwerty8224, 2016-09-21:
Right now I'll check)))

donovan6000, 2016-09-21:
I just realized that version of M33 Manager doesn't have a button that lets you switch to bootloader mode. M33 Manager [V0.05](https://github.com/donovan6000/iMe/archive/V00.00.00.09.zip) was the first version that has that button.

qwerty8224, 2016-09-21:
ок check now!

qwerty8224, 2016-09-21:
V0.05 there is an error

qwerty8224, 2016-09-21:
V 0.05 there is an error

donovan6000, 2016-09-22:
I still have no idea what's causing the problem. Can you try [every version of M33 Manager](https://github.com/donovan6000/iMe/releases) and let me know which versions don't cause this issue?

qwerty8224, 2016-09-22:
OK . How to check'll let you know.

closed 2017-02-18 by qwerty8224