donovan6000/iMe #50: Building iMe, .hex to big

MaxTimothy, 2017-02-16:
I'm attempting to add acceleration to iMe, but when I attempt to install the newly made hex using m33 I receive an error "is too big" in the dialog box. My project's firmware hex is 34kb, I did notice that the hex that comes with m33 is only 31.5kb. If I build the unmodified project it produces a 33.9kb hex, is there something that I need to change in atmel studio for it to produce a smaller hex?

donovan6000, 2017-02-16:
Make sure your using the latest version of Atmel Studio and that you have it setup to use the [provided Makefile](https://github.com/donovan6000/iMe/blob/master/Makefile). iMe's [compiler settings](https://github.com/donovan6000/iMe/blob/master/iMe.cppproj#L281-L286) already has it configured to to that, so you should be able to open the [project file](https://github.com/donovan6000/iMe/blob/master/iMe.atsln) with Atmel Studio and be able to compile it to a small enough size.

MaxTimothy, 2017-02-20:
I'm on version 7.0.1188, have the project configured to use the iMe makefile. Still can't compile the unedited iMe master into a hex file smaller than 33.9kb. Any more suggestions?

donovan6000, 2017-02-23:
It looks like the Linux and Windows compilers are producing different results even though they're the same versions. The compiler options that are causing this are the linker time optimization settings `-flto` and `-flto-partition`. Unfortunately I'm not able to produce a small enough firmware using the Window's compiler without those settings working correctly.

Here's the resulting firmware size in bytes when using different linker time optimization settings on the different operating systems.
```
Linux
	-fno-lto: 33318
	-flto-partition=1to1: 32256
	-flto-partition=none: 32502
	-flto-partition=max: 32758
	-flto-partition=balanced: 32504

Windows
	-fno-lto: 33318
	-flto-partition=1to1: 34812
	-flto-partition=none: 35042
	-flto-partition=max: 35086
	-flto-partition=balanced: 34916
```

MaxTimothy, 2017-02-23:
I was wondering if that was the issue, I'll just compile it on Linux then. Thanks!

closed 2017-02-23 by MaxTimothy