donovan6000/iMe #51: Printer stuck on "positioning extruder" when trying to unload filament

Nold360, 2017-03-25:
Hi,

today I'm having some issues with iMe + M33-Fio on OctoPrint.
When I'm trying to unload the filament, Octoprint stucks at "Positioning extruder".

Here is a snippet from the octoprint.log:
```
| Send: N0 M110*35
| Recv: ok 0
| Send: N1 G90*17
| Recv: ok 1
| Send: N2 G0 Z15 F90*58
| Send: N3 G28*16
| Send: N4 G4*41
| Send: N5 M65536
| Recv: ok 2
| Send: M105
| Send: M105
| Recv: ok 3
| Recv: ok 4
| Recv: rs 5
| Printer requested line 5 but no sufficient history is available, can't resend
```

It's not much information, but all I can get ATM. :(

Yorper, 2017-03-26:
Are you using version 1.3.2 of Octoprint? If so then that is why you can not connect the printer. Follow the link below for a solution until M33-Fio is officially updated.

https://github.com/donovan6000/M33-Fio/issues/216

donovan6000, 2017-08-09:
It's been a while since anyone has responded to this issue, so I'll probably close it in a couple of days. Let me know if you still have any interest in testing/fixing this issue.

closed 2017-08-12 by donovan6000