donovan6000/iMe #56: Extruder does not move with iMe 1.24

alancnet, 2017-06-13:
I couldn't imagine I'm the only one with this issue, but I don't see anyone else reporting it. I suppose I must.  When I print the test border, the extruder motor will move. When I print from something sliced, it won't .. If I install M3D firmware, it works.

donovan6000, 2017-06-15:
I'm not really sure what's causing this. Can you verify that your Settings > M33 Fio > 'E motor steps/mm' setting is at 128.451375 steps/mm?

alancnet, 2017-06-21:
Almost... 128.45137

Is there a way to reset all the settings to default?

donovan6000, 2017-06-22:
128.45137 is probably the closest single precision floating point number to 128.451375, so that's fine.

All the default settings for M33 Fio and the printer can be found [here](https://github.com/donovan6000/M33-Fio/blob/master/octoprint_m33fio/__init__.py#L2118-L2176). I wouldn't recommend changing any of the [first few though](https://github.com/donovan6000/M33-Fio/blob/master/octoprint_m33fio/__init__.py#L2118-L2129) since you'll have to recalibrate your printer's bed and backlash if you do.

donovan6000, 2017-08-09:
It's been a while since anyone has responded to this issue, so I'll probably close it in a couple of days. Let me know if you still have any interest in testing/fixing this issue.

closed 2017-08-12 by donovan6000