donovan6000/iMe #61: M3D Micro+

blandman, 2017-10-12:
I noticed there was no request here for iMe support for the Micro+, but I found this [thingiverse thread](https://www.thingiverse.com/groups/m3d-the-micro-3d-printer/topic:19755).

Is this still being worked on? 

Is there any assistance I could offer?

Thank you for such a great service, we really want use it on our Micro+!

mdikov, 2017-10-26:
Yep same here, just got a Micro+

Rufuskumbowii, 2018-03-11:
I just upgraded my micro to micro +. the plus is much better than the micro, so awesome there, but all my 3rd party stuff no longer works.  

How can I help make this happen.  I genuinely just started liking my printer--I figured I had wasted money and thus incurred stupid tax on 3d printers, until I found this software and simplify3d.  

I was just getting some prints to come out almost reasonably well (extremely high failure rate with just the micro OEM) and decided to try something else.  so thank you for awesomeness!! and I hope to be able to use this for my micro plus.      if I can do anything to help please let me know.

THansen123, 2018-12-11:
How can we move forward?

Blueforcer, 2020-09-18:
I think the micro+ support is dead?!