donovan6000/iMe #62: Seg fault when running M33 Manager on Ubuntu Linux 17.10

ianc1215, 2017-11-23:
I am having an issue running the M33 manager on my Ubuntu 17.10 machine. The program won't even start. If I run it from terminal I see this (see picture).

The error does not give me much to report. My machine does not have anything special installed. I only have a basic Ubuntu installation.

Any idea what the issue could be? If you need me to test anything I can do so.

[image]

donovan6000, 2017-11-27:
For some reason the Mir display library that Ubuntu 17.10 comes with doesn't like that I statically linked some libraries. If you remove the `-static-libgcc -static-libstdc++` in [M33 Manager's Makefile](https://github.com/donovan6000/iMe/blob/master/M33%20Manager/Makefile#L87) then you'll be able to compile a version that works in Ubuntu 17.10.

ianc1215, 2017-12-08:
Awesome! Thanks, I will give that a try and let you know.