donovan6000/iMe #63: Printer occasionally gets lost

cynr1c, 2017-11-24:
Hi,
first of all, thank you for iMe and M33Fio! I've come to really like this combination on my Micro, especially the print quality due to the different approach to backlash compensation is a big plus for me!

Now to the issue I am experiencing (1.0.0.25): Rarely the printer gets lost, meaning an axis seems to move a full step too far, causing a print to split somewhere in between. It just starts extruding a few mm too far to the right or back (I've had both). This problem only occurs sporadically; I've had it 3 or 4 times in the last 2 or 3 months. 
Just now it occurred while printing the first layer of a raft, there you can see the problem pretty clearly. After the 5th line in the Y direction the printer skipped roughly 4mm and kept this offset for the remaining print until I aborted it. Image is oriented as the part was in the printer, bottom being the device front.
[image]

I know sporadically occuring bugs are a b*tch to fix, but maybe you can look into this if you've got the time..
Thank you!

donovan6000, 2017-11-27:
Can you send me the G-code file that you were printing when this happened. And can you also send me your printer's settings. You can get them by clicking the `Save printer settings to file` button toward the bottom of the Calibration section in OctoPrint's Control tab.
[image]

And did you experience this problem when you printed the file again?

cynr1c, 2017-11-27:
Sure, here you go. Printer settings have changed a little since the export (bed calibration Z0 offsets of corners and center), cannot access the machine right now. And no, the print came out just fine from the same Gcode after scraping off the raft and starting again, so it does not look like a Cura problem..

[Wpc_drop_target_r01.zip](https://github.com/donovan6000/iMe/files/1506630/Wpc_drop_target_r01.zip)
[Printer Settings 8.11.2017.zip](https://github.com/donovan6000/iMe/files/1506635/Printer.Settings.8.11.2017.zip)

akohlsmith, 2018-01-02:
Not sure if this is related, but I've noticed that my Micro will occasionally "forget" to bring Z down to the bed at the start of the print. With Octoprint the normal start of print looks like this:

* extruder is moved to the front left corner
* extruder brought up to temp
* extruder moved down (Z) about 5mm above bed and a blob of filament is actively extruded
* extruder will then move down to bed and to the right, "kiss" the bed and then rise (Z) up about 50mm  (this leaves a "stick" of filament from the bed to the extruder nozzle

Normally at this point it will wait about 10-15s then start the print, moving to position while coming down to the bed. The printer will sometimes "forget" to move the extruder down to the bed and the print will start up in the air.