donovan6000/iMe #67: No safty features

guysoft, 2018-11-18:
Hey,
Does the firmware have no safety features such as thermal runaway and max-temp?
I see no documentation on that.

donovan6000, 2018-11-19:
iMe will [prevent you from setting the extruder's temperature](https://github.com/donovan6000/iMe/blob/master/main.cpp#L1274) above [285](https://github.com/donovan6000/iMe/blob/master/heater.h#L10) degrees Celsius.

It will also [turn off the fan, led, heater, and motors after 10 minutes of inactivity](https://github.com/donovan6000/iMe/blob/master/main.cpp#L227-L234). The devel branch [fixed an issue](https://github.com/donovan6000/iMe/commit/029afa5c1c363e373a08d53bb8662589d50f3d67) with this where the command that checks what the temperature is, M105, was resetting the inactivity counter.

donovan6000, 2018-11-20:
Ok, then I agree that iMe should be in the list if insecure firmwares. Can you provide this M115 response to foosel so that she can add it to the list? `PROTOCOL:RepRap FIRMWARE_NAME:iMe FIRMWARE_VERSION:00.00.01.26 MACHINE_TYPE:Micro_3D EXTRUDER_COUNT:1 SERIAL_NUMBER:0000000000000000`

From my own experience, I can also confirm that the Micro 3D printer's extruder circuity provides very little to mitigate thermal runaway. During iMe's development, I once accidentally heated up the extruder to the point where something in the extruder's enclosure started to smoke. So potentially the Micro 3D printer could experience thermal runaway if the heating element moves out of place or if the firmware crashes while the it's heating the extruder.

guysoft, 2018-11-20:
cc @fooel .
I am sick today and my M3D is at work, can confirm then.

guysoft, 2018-11-20:
@donovan6000 I think firmware crashes are also possible on Marlin to cause a thermal runaway. The question is if its possible to add it to the firmware?

foosel, 2018-11-20:
Just added detection of this, will be in 1.3.10rc3+

guysoft, 2018-11-20:
Happy its known, would be great if we can fix it.
also might be a good idea to warn on stock firmware.

donovan6000, 2018-11-24:
What do you think is a good way to detect thermal runaway? It seems like detecting this on the Micro 3D printer is kinda tricky since the printer's heating element is also what is used to measure the extruder's temperature.

If the heating element moves out of place while the extruder is already hot, then the temperature readings would probably increase a bit since the extruder would no longer be acting like a heat sink for the heating element. But the readings would stabilize shorty after that. I've seen temperature readings on the Micro 3D printer continually vary by more than 15 degrees from a set temperature under normal circumstances, so this seems difficult to differentiate thermal runaway from normal behaviour in this case.

We'd also have to account for if the heating element was already moved out of place before we sent the printer a command to start heating up the extruder. In that case, I think the temperature readings would still look normal since the heating element would still gradually heat up, just quicker than usual. This would also be difficulty to differentiate from normal behaviour since each Micro 3D printer has some unique temperature calibration values that are used to determine the extruder's temperature, and this causes the temperature curve when the extruder is heating up to vary from printer to printer. So some printer will already appear to heat up quicker than others.

guysoft, 2018-11-25:
@foosel 
Stock firmware replies with this:
``Recv: ok REPETIER_PROTOCOL:2 FIRMWARE_NAME:Micro3D FIRMWARE_VERSION:2016040401``
Its also unsafe (likely much worse than @donovan6000 's image , we don't know).

Also, if you have any advice for https://github.com/donovan6000/iMe/issues/67#issuecomment-441346719  (even who might have an answer). That would help!

foosel, 2018-11-26:
I've added the stock firmware to the printer safety plugin as well, will be detected starting with 1.3.10rc3.

I have no advice on the challenge of reliably detecting a thermal runaway here I fear. Other firmwares do it by checking temperature increase against applied power to the heater as far as I know. Too big of an increase with now power -> error. Too little of an increase with power -> error. Sounds like the design of the Micro makes this tricky though.