donovan6000/iMe #70: Z axis shifts during printing

john-junhong, 2020-04-21:
Hi! First of all, thank you very much for your excellent firmware! I am using MatterControl to control the M3D printer. However, I noticed that the z axis of the printer always slowly shifts down about 3mm during the printing. This annoying problem results in a shorter and contractive printed model.
[image]

I also investigated a little bit. I exported the GCODE and did some test. First of all, I preset MatterControl with the right speed limit, and make sure the feed rate in GCODE is under 960mm/min.
1. X speed limit 1500mm/min
2. Y speed limit 1500mm/min
3. Z speed limit 60mm/min
4. E extruding speed limit 102mm/min
5. E retracting speed limit 360mm/min

- Before printing, the printer is calibrated by G30, the bed position is valid. If I move to position z=0, the extruder hits the bed exactly. After printing, the bed position is still "valid". However, if I move to position z=3mm, the extruder would hit the bed, which means the extruder shifts down about 3mm.
- There are three kind of move commands in the GCODE: G1 X,Y; G1 X,Y,Z; G1 Z. I keep all the commands in the GCODE, After printing, the shifting happened.
- As a test, I deleted all the G1 X,Y and G1 X,Y,Z commands, and kept the G1 Z commands. There wasn't shifting happened.
- As a test, I deleted all the G1 Z and G1 X,Y,Z commands and kept the G1 X,Y commands. There wasn't shifting happened.
- As a test, I only kept the G1 X,Y,Z commands. There wasn't shifting happened.

I also walked through the code and didn't find places that may cause this problem. I was very confused. Could you please tell me do you have any idea where this problem may come from? I am willing to share the GCODE and print setting with you. Thank you very much!

dachew, 2020-04-21:
I can't speak for these printers anymore - but when this happens on my current printer it's due to lose hardware: bed, gantry, etc.,  On the M3D printers, I had trouble with the print bed moving around during the print. Most likely, it's not a GCODE problem you have there. As the printer rises up, the print can not move underneath. Z shift is almost always some sort of physical problem because it's caused by the part moving during print. If you are printing a tall part, try adding support structures to the sides, sort of like beams leaning up against it.

dachew, 2020-04-21:
Can you post a picture of how this part looks in the printer?

john-junhong, 2020-04-22:
> Can you post a picture of how this part looks in the printer?

Thank you very much for your reply! Today I changed to use another software named Repetier and the embedded Cura slicer which were tested by donovan6000. However, the Z shift still happened. I am wondering it must be something related to the hardware. Here is the picture of the part I just printed tonight. 
[image]

It's very suspicious, the Z shift always happens near the bottom. It happens *all the time*! And I think it's not related to the parts. Because, in all the tests I have done before, I stop the hotend and only move the extruder in the air, the Z shifts still happen! 

And as I stated before, only under the particular circumstances, like running the generated GCODE in original sequence will it happen. If I only keep the Z move command, there won't be Z shift, which should *prove that the bed and gantry are normal*. It's very bizarre.

dachew, 2020-04-23:
yep, same thing would happen in my M3D Pro. I had set some wedges up to jam the bed into place, which helped a lot.