donovan6000/iMe #71: Cannot connect to M3D Micro on MacOS Big Sur

IhorSyerkov, 2021-06-12:
Application doesnt see printer connected via USB. 
In **Serial Port** menu only one option `Auto`.
  (Windows veriosn works fine)

M33 Manager V0.16
OS: MacOS Bir Sur 11.4

Thanks :)

majorcode, 2021-06-15:
Any clue on how to fix this? I'm having the same problem.

IhorSyerkov, 2021-06-16:
I don't know cpp. But I looks like `void Printer::updateAvailableSerialPorts()` doesn't fill `availableSerialPorts` vector. 
 Priter device actually appears as `/dev/tty.usbmodem14501` device. I'm not sure is it suppose to be **usbmodem** or **usbserial**

IhorSyerkov, 2021-06-17:
@majorcode good news. I was able to fix it. The only problem that I didn’t manage to create dmg package. You can try to pull my branch build it for your self. I hope @donovan6000 will help us with this final step :)

metawops, 2021-07-11:
I have the same problem with my M3D Micro and the M33 manager. I'm using macOS Catalina (10.15.7) and the plugged-in M3D Micro connects as `/dev/tty.usbmodem14101` but doesn't show up in M33 Manager. 🤷‍♂️ 
I will try to get your fix working, @IhorSyerkov ...

metawops, 2021-07-11:
Oh no, it's dependant on wxWidgets which I don't have installed, obviously. Oh my. 😞

metawops, 2021-07-11:
YEAH! I was successful! Had to download the wxWidgets source and compile it and afterwards installed it according to the OS X install.md file. Then I successfully built the M33 Manager from your branch, @IhorSyerkov (which gave several warnings I ignored). I was looking for the M33 Manager Icon but it's not there. Instead you just have to double click the created `M33 Manager` file that has the default Terminal icon. It launches a Terminal which, in turn, launches the M33 Manager GUI app. It shows as a default folder icon in the Dock. 
Even better, the `/dev/tty.usbmodem14101` entry showed up in the "Serial Port" dropdown list and my M3D Micro got detected. I was boldly able to flash its firmware to the iMe firmware, too. 🤩 
Afterwards, the M33 Manager app did a bed position calibration succesfully.
Awesome work, @donovan6000!! 👍👍 
Now let's get back to OctoPrint and see if that helps with the M33 Fio plugin. [Here's a discussion about that](https://community.octoprint.org/t/need-to-install-older-version-to-run-m33-fio-for-micro-m3d/19290) in case you're interested.

solidfox, 2022-12-12:
The changes by @IhorSyerkov helped for me but compiling on Ventura was a bit of a pain. Updated the Makefile accordingly (now only requires you to `brew install wxwidgets` and added an updated dmg [here](https://github.com/solidfox/ime).